0xD34D/klipper_ender3_v3_se #1: First print works fine, subsequent prints miscalculate the offset causing it to be higher off the bed

0xD34D, 2023-11-14:
**Describe the bug**
After a successful print using PRTOUCH_CHECK_ZOFFSET, the next print to start that uses this will be too high.

**To Reproduce**
Steps to reproduce the behavior:
1. Start and complete a print using PRTOUCH_CHECK_ZOFFSET in start of print
2. Start another print using PRTOUCH_CHECK_ZOFFSET in start of print
3. Watch filament be deposited too high from the bed :/

**Expected behavior**
First layer is properly adhering to the bed

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

pwlgrzs, 2023-11-14:
That's interesting because I did not experienced this. Happened after recent commits (on 14.11)?

0xD34D, 2023-11-15:
@pwlgrzs I think I've had this problem since the beginning.  It could be my start and end macros for all I know if I'm the only one experiencing this problem.  I also noticed my host and mcu are out of sync.  That's from me experimenting so I'll need to fix that and rule that variable out.

0xD34D, 2023-11-15:
The issue appears to be with having a bed mesh applied when probing.  If `BED_MESH_CLEAR` is called first then there is no issue with subsequent prints.  Will need to figure out if there is a graceful way to handle this or if it's just gonna be required to clear the bed mesh prior to probing.

FinalX1992, 2023-11-20:
so a Macro like this should work Correct?
BED_MESH_CLEAR
PRTOUCH_PROBE_ZOFFSET
BED_MESH_PROFILE LOAD="default"

0xD34D, 2023-11-20:
> so a Macro like this should work Correct?
> BED_MESH_CLEAR
> PRTOUCH_PROBE_ZOFFSET
> BED_MESH_PROFILE LOAD="default"

That's the order I have mine set in my PRINT_START macro, should be good to go with that.

shebang79, 2024-01-15:
I'm struggling with this myself, I thought this might be the problem I've been struggling with, but no. No matter what I do, the auto-z results in air printing. I'm at a loss here, I've spent weeks trying everything I could think of, but always the exact same results -- the nozzle is way too high.

imranh101, 2024-03-05:
Hey, still having this happen... I tried adding the bedmesh macro you have above in to my Cura start g-code but still not working.  All sorts of people here discussing what they use for their macro that works but no one has copied and pasted a working one :( Anyone help me out? 

`M104 S150 ;Heat hotend, no wait
M190 S{material_bed_temperature_layer_0} ;Heat bed to Cura and wait

M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

M420 S1; Enable mesh leveling

BED_MESH_CLEAR
PRTOUCH_PROBE_ZOFFSET
BED_MESH_PROFILE LOAD="default"

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
M109 S{material_print_temperature_layer_0}
G1 X10.1 Y145.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y145.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0  ;Reset Extruder
G1 E-1.0000 F1800 ;Retract a bit
G1 Z2.0 F3000 ;Move Z Axis up
G1 E0.0000 F1800`

imranh101, 2024-03-05:
Well, was able to overcome the above. I read another message about how calling the prtouch probe offest in your start code is bad and rearranged and got over that. im having other print quality issues now but i also kind of want to chalk it up to the stock creality bed. never been a fan of their spring steel. i ordered a pei one i had good experience with in the past and gonna continue to tweak settings and try to get a good print going. Thanks for this build of Klipper, I was out of printing for 6 months cuz got PO'd at my original Ender 3 and as soon as I saw the SE with all its features, I knew I wanted it *if* I could put Klipper on it. And this page was my answer.

imranh101, 2024-03-06:
Okay, I seem to have overcome above issues with a lot more calibration. However I am still seemingly having issues w/ Z-Offset. I printed something last night just fine, without any printer restart or anything, I cleared print bed and fired off another print and nothing sticks, z-offset is wrong. Here is my start G-code, maybe you can tell me if I am doing something wrong, or perhaps it has nothing to do with my Start Gcode?

`M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

BED_MESH_PROFILE LOAD=default

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
M109 S{material_print_temperature_layer_0}    
G1 X10.1 Y145.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y145.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0  ;Reset Extruder
G1 E-1.0000 F1800 ;Retract a bit
G1 Z2.0 F3000 ;Move Z Axis up
G1 E0.0000 F1800`

imranh101, 2024-03-06:
This still has to be related in some way to bed mesh loading. I did a bedmeshclear, then z-probe-offset, confirmed w/ paper test good gap, and then loaded bed mesh again, tried paper test and no grip at all. Why would loading the bedmesh cause probe offset to go wrong if I made no changes zoffset? I'm so lost and so unable to print.

imranh101, 2024-03-07:
> I use prusaslicer so my start code won't work. I fried my cr touch so I made a klicky probe and a stop and use protoloft for my auto zheight now, and I love it. And it looks cool so+5 Cool score. Anyway, in his wiki there's this to help with this.
> 
> "process may fail after auto-calibration if the applied offset is too high. However, there is a workaround by setting the GCode offset back to zero within the attach and dock macros. To make it valid only in the macro, you can use the save/restore GCode state commands like this:
> 
> SAVE_GCODE_STATE NAME=YOUR_MACRO_NAME SET_GCODE_OFFSET Z=0.0
> 
> your macro content here... (This would b like bedmesh and ztilt)
> 
> RESTORE_GCODE_STATE NAME=YOUR_MACRO_NAME This will save the state including the calibrated offset, reset it to zero, run the attach/dock routine and restore the calibrated offset."

Not gonna lie, I'm a little lost trying to figure out the macro part mentioned here 😟 Do you have a link to this wiki page? I'm not finding it and perhaps more context would help. Thanks for your response.

imranh101, 2024-03-07:
I think I've got it all dialed in now. I realized my z gantry was "loose" enough that during fast Y movements it would sway enough to change the z-distance enough to mess up the print. Wow! I flipped it on it's side, cranked down the two sets of 3 bolts that hold the upper frame to the lower frame (used a high quality tool - Chinesium strips out on these bolts way too easy) and torqued it down till I felt it was real secure and then flipped it over and boom - I was able to do an auto probe offset, hit print and it started going down fine. I'm currently at 200mm/s in slicer and 120% in Mainsail and going swell. I also did a input shaping calibration before, using the $12 bigtreetech usb-c accelerometer and it went great, but I should probably do that once more now that I've tightened the gantry down - I'm sure that change the resonant frequencies. Thank you for this build/upload!

derekmcghee, 2024-03-07:
https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Use-It#command-calibrate_z

I added a 5th stepper and use z-tilt 🤔🫣

ForbiddenEra, 2025-11-11:
> https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Use-It#command-calibrate_z
> 
> I added a 5th stepper and use z-tilt 🤔🫣

@derekmcghee still want to know how you're powering that? :D