0xD34D/klipper_ender3_v3_se #2: Throws a "Move out of range error"

pi3123, 2023-11-16:
**Describe the bug**
Move out of range: 7.732 48.634 -3.452 [0.000]
Move out of range: 7.732 48.634 -3.452 [0.000]

Throws this error when running the PRTOUCH_PROBE_ZOFFSET

**To Reproduce**
Steps to reproduce the behavior:
1. Run the PRTOUCH_PROBE_ZOFFSET

**Expected behavior**
Get z_offset

0xD34D, 2023-11-16:
@pi3123 You might need to adjust your `position_min` under the `[stepper_z]` section.  I have mine set to `-5` to ensure I have enough room to go negative when probing for the z-offset of the nozzle.  I think something in the range of -3.5 to -4.0 should be more than enough.

pi3123, 2023-11-17:
That works! But now the z offset is much higher than the perfect manual one I got. I got a 0.735 and its giving me 0.913. Anything I gotta configure?

pwlgrzs, 2023-11-28:
> That works! But now the z offset is much higher than the perfect manual one I got. I got a 0.735 and its giving me 0.913. Anything I gotta configure?

Not really, my experience is that z-offset with this version of prtouch is "good enough" but not perfect and definitely lower (and there's no code publicly available to use newer version).

For example my auto z-offset is 1.982 but Benchy bottom letters are barely visible. If I change to 1.765 letters are much better.

So there're three options:

- use auto Z and don't change, most prints will be OK provided you don't get completely bad value,
- use auto Z to measure and then correct it until you find a perfect one,
- use auto Z and compensate in g-code by small abount