0xD34D/klipper_ender3_v3_se #3: Printer not connecting

Samywamy10, 2023-11-16:
**Describe the bug**
After making the firmware using this repo with commit `cdc11ba5`, Klipper on my rpi won't connect to my printer, with the following error:
```
serialhdl.error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/sam/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/sam/klipper/klippy/serialhdl.py", line 261, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/sam/klipper/klippy/serialhdl.py", line 318, in get_response
    self.serial.raw_send_wait_ack(cmds[-1], minclock, reqclock,
  File "/home/sam/klipper/klippy/serialhdl.py", line 253, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/sam/klipper/klippy/serialhdl.py", line 61, in _error
    raise error(self.warn_prefix + (msg % params))
serialhdl.error: mcu 'mcu': Serial connection closed
```

But when I flash back with the original Klipper firmware its fine. I tried renaming the file to be different e.g. `klipper1.bin` but didn't seem to make a difference

Working binary:

[klipper.bin.zip](https://github.com/0xD34D/klipper_ender3_v3_se/files/13377865/klipper.bin.zip)

Not working:

[klipper.bin.zip](https://github.com/0xD34D/klipper_ender3_v3_se/files/13377867/klipper.bin.zip)

Not sure if comparing binaries is possible but not sure what else to provide. Any ideas?

0xD34D, 2023-11-16:
Hey @Samywamy10,

Did you run `make menuconfig` and setup the MCU before building the binary?  If so, what settings did you use?

If everything is correct with the configuration then you might try `make clean && make` to get a clean build then reflash.  As always, just make sure the `.bin` is a different name than the last time you flashed ;)

Samywamy10, 2023-11-16:
Hey yeah I did the `menuconfig` steps (I was familiar as already did it to generate the original klipper bin file). I used the settings described in `printer.cfg`

I tried deleting the `out` folder and re-compiling from scratch after switching to your repo. I renamed the bin file on the SD card to be something different.

Samywamy10, 2023-11-19:
I've actually given it another try and still no luck. I'm naming it `clipper.bin` and still Klipper isn't connecting, just getting the same
> Printer is not ready
The klippy host software is attempting to connect. Please
retry in a few moments.

As soon as I switch back to the original `klipper.bin` it works.

I'm on your latest commit `eb5d6b12` and built from this commit too. These are my `menuconfig` settings:

[image]

Samywamy10, 2023-11-19:
So I re-read my message here and the [printer.cfg](https://github.com/0xD34D/ender3-v3-se-klipper-config/blob/main/printer-creality-ender3-v3-se-2023.cfg) and realised I'm an idiot and had the wrong communication interface.

PEBCAK - keen to try your build!

closed 2023-11-19 by Samywamy10