0xD34D/klipper_ender3_v3_se #4: Z_Adjust doesn't save when restarting printer

gadgethd, 2023-11-18:
**Describe the bug**
Z_Adjust doesnt apply when restarting printer as it is not a saved value by Klipper, it seems.

**To Reproduce**
Steps to reproduce the behavior:
1. Do a Z auto calibrate
2. Save Config and restart
3. see value gets removed after restart

Before restart:

[image]

After restart:

[image]

**Expected behavior**
Whole Z offset to be saved on restart

**Additional context**

I believe this could be fixed by changing line 834 to include nozzle_z_offset in the calculation, then it would save as if it were a regular offset.

pi3123, 2023-11-18:
The ZOffset is saved to the printer.cfg under [bltouch]
It doesn't save to the regular offset but it definetily saves it after running the SAVE_CONFIG

gadgethd, 2023-11-18:
Yes the normal Z offset is saved, however the Z_Adjust doesnt get saved as its in the temp Z offset.

gadgethd, 2023-11-18:
Ok so ive figured out why its doing this, its quite smart and I didn't notice. The temporary value is the difference between the previous Auto Z Offset and the new one, so you don't have to restart the printer. I think this is what you were mentioning before and I didn't fully understand it.

closed 2023-11-18 by gadgethd