0xD34D/klipper_ender3_v3_se #6: MCU Protocol error

VeGoVeVO, 2023-11-27:
**Describe the bug**
after replacing klipper firmware with yours I get this error which doesn't let me connect to the printer using an older version of Mcu here's the error screen I hope you can I help me I'm trying to get the auto z offset to work since it started to piss me off without it since I don't have a lot of time to constantly do it if I do a few tweaks to the printer thanks in a advance and good job on the progress so far.
[image]

pwlgrzs, 2023-11-27:
Did you install both firmware and klipper service from this repo?

VeGoVeVO, 2023-11-27:
> Did you install both firmware and klipper service from this repo?

i did this "git clone https://github.com/0xD34D/klipper_ender3_v3_se then changed the name from klipper_ender3_v3_se then ./klipper/scripts/install-octopi.sh" to klipper after that "cd ~/klipper/
""make menuconfig" followed the config "STM32F103 with a "28KiB bootloader" and serial (on USART1 PA10/PA9)" then "Flash this firmware by copying "out/klipper.bin" to a SD card and
# turning on the printer with the card inserted" and that's it let me know if I missed something I know little about these issues so I would accept any help given thank you.

pwlgrzs, 2023-11-27:
Okay, I will give you what worked for me and what I have documented.

First start with this:
https://github.com/pwlgrzs/awesome-resources-for-ender-3-v3-se/blob/main/remote-control/klipper/README.md#install-instructions-for-0xd34d-klipper-fork
Then follow with this:
https://github.com/pwlgrzs/awesome-resources-for-ender-3-v3-se/blob/main/remote-control/klipper/README.md#installation-and-configuration

I hope @0xD34D does not mind me providing links to my documents, but I tried to make clear documentation for people who would like to use his work.

pwlgrzs, 2023-11-28:
On the screenshot "Your klipper version" is same as mine, so that's good I guess, I wonder why your klipper MCU is still different. Perhaps you could try renaming bin file made with this repo to different name (also there should not be any other bin files on the sdcard) and see if this flashes it right? Alternatively, flash stock and retry with klipper bin.

VeGoVeVO, 2023-11-28:
> On the screenshot "Your klipper version" is same as mine, so that's good I guess, I wonder why your klipper MCU is still different. Perhaps you could try renaming bin file made with this repo to different name (also there should not be any other bin files on the sdcard) and see if this flashes it right? Alternatively, flash stock and retry with klipper bin (maybe also renamed).

renaming the file on the sdcard worked thanks a lot ^^.

closed 2023-11-28 by VeGoVeVO

pwlgrzs, 2023-11-28:
That's great!

0xD34D, 2023-11-28:
> > I hope @0xD34D does not mind me providing links to my documents, but I tried to make clear documentation for people who would like to use his work.

I don't mind at all.  I've been slacking off when it comes to documentation 🤪