0xD34D/klipper_ender3_v3_se #8: how to install

matnorte, 2023-11-30:
put a tutorial to how to install using this klipper config

pwlgrzs, 2023-11-30:
Hello @matnorte, you can follow 2 guides here https://github.com/pwlgrzs/awesome-resources-for-ender-3-v3-se/blob/main/remote-control/klipper/README.md

freeman5, 2024-01-08:
why when i do make menuconfig i dont see Micro-controller architecture: STMicroelectronics STM32 ?

freeman5, 2024-01-10:
I have a problem, i change the repo in kiauh, i do the installation, i do make menuconfig, and make, i take the klipper.bin but now i have problem with MCU, because the firmware is different version of klipper o_O

Forrest-O, 2024-03-10:
I used [KIAUH](https://github.com/dw-0/kiauh)'s custom repo thing and a [YouTube tutorial](https://youtu.be/LrBiwabN-Y8?si=ed4jtgXjgRoHZkhk) (turn on subtitles for english).