0xD34D/klipper_ender3_v3_se #10: I got error while compiling klipper.

HongyongChoi7795, 2024-01-12:
While I compile klipper firmware for rp2040, I got this error.

[image]

My config was this.
[image]

Alexandrinn, 2024-01-12:
# This file contains pin mappings for the stock 2022 Creality Ender 3
# V3 SE. To use this config, during "make menuconfig" select the
# STM32F103 with a "28KiB bootloader" and serial (on USART1 PA10/PA9)
# communication.
# "Enable extra low-leverl configuration options" and "Disable SWD on startup"
[image]

HongyongChoi7795, 2024-01-12:
> "# This file contains pin mappings for the stock 2022 Creality Ender 3 "# V3 SE. To use this config, during "make menuconfig" select the "# STM32F103 with a "28KiB bootloader" and serial (on USART1 PA10/PA9) "# communication. "# "Enable extra low-leverl configuration options" and "Disable SWD on startup" ![Screenshot 2024-01-12 230411](https://private-user-images.githubusercontent.com/33482807/296365250-84256044-81bf-47bd-8a63-001058991993.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUxMDI3MTIsIm5iZiI6MTcwNTEwMjQxMiwicGF0aCI6Ii8zMzQ4MjgwNy8yOTYzNjUyNTAtODQyNTYwNDQtODFiZi00N2JkLThhNjMtMDAxMDU4OTkxOTkzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEyVDIzMzMzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgxMWVjYTFkMTVjY2ZjYmJiMDM1ZWIzMGYyMTMxNTIzY2IzNDJiYmZkODJiNWFmN2EyMmFkZWY3MWRhNzM1YzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VweCa9OeNcJuOYiL_95gpPhn0vUe_pAwzgTZTnORCsA)

I tried to make firmware for my portable adxl345.

schnoog, 2024-01-21:
As temporary fix to be able to compile the firmware for the RP2040 I added
 
#define CONFIG_CLOCK_FREQ 12000000

to src/hx711s.c and  src/dirzctl.c
would be great if that could be avoided.

commit 21eb915d (2024-01-31): Merge pull request #11 from schnoog/master

fix #10 - Compilation fails on other architectures than STM32

Verified compiles, and all functions work normally on E3V3SE with this change.  Thank you @schnoog !
commit b41a93be (2024-01-31): fix #10
Added #include "autoconf.h" so src/dirzctl.c and src/hx711s.c
Without it CONFIG_CLOCK_FREQ is unknown to both files and compiling
of additional MCUs (f.e. RP2040) fail.
Changes tested with V3SE (STM32) and RP2040 as micro-controller architecture.
closed 2024-01-31 by 0xD34D