0xD34D/klipper_ender3_v3_se #12: AIR Printing

VeGoVeVO, 2024-01-27:
after updating everything and probing the printer is now air printing and when cleaning the nozzle at the beginning of the print its not even on the bed so i think theres problems with calculations done or something because it worked fine until the latest update i hope you can help me fix it cause i like this a lot.

VeGoVeVO, 2024-01-30:
i tried everything even changing zoffset doesnt work because its like the change doesnt get applied evern when its changed in the printer.cfg please someone with an idea holatme

pwlgrzs, 2024-01-30:
Could you please provide a copy of your printer.cfg before and after running an automatic z-offset?

VeGoVeVO, 2024-01-30:
Yes for sure 
[Printer After.txt](https://github.com/0xD34D/klipper_ender3_v3_se/files/14103765/Printer.After.txt)
[Printer before.txt](https://github.com/0xD34D/klipper_ender3_v3_se/files/14103766/Printer.before.txt)

naveennisad, 2024-01-30:
I dont think it updates offset value in config file

VeGoVeVO, 2024-02-04:
anyone figured this thing out im literally stuck idk if i should reinstall all or what to do but yeah im currently frozen

pwlgrzs, 2024-02-04:
Clearly your before/after z-offset is the same.

- when running PRTOUCH_PROBE_ZOFFSET is the printer doing anything at all?
- if above is true, do you run SAVE_CONFIG after process is completed?

VeGoVeVO, 2024-02-04:
> Clearly your before/after z-offset is the same.
> 
> * when running PRTOUCH_PROBE_ZOFFSET is the printer doing anything at all?
> * if above is true, do you run SAVE_CONFIG after process is completed?

it does move above the sensor and uses prtouch then the nozzle to calculate but i rarely save the config because i thought since my rpi is running 24/7 its not needed
but i will try it with a save config and let you know in a few minutes

VeGoVeVO, 2024-02-04:
> Clearly your before/after z-offset is the same.
> 
> * when running PRTOUCH_PROBE_ZOFFSET is the printer doing anything at all?
> * if above is true, do you run SAVE_CONFIG after process is completed?

i tried saving it insta and the zoffset is like way to close to the bed but at least it worked i feel so stupid now thanks a lot for the help i will just mess around with the settings to get it at the right z

pwlgrzs, 2024-02-04:
Okay, so now if you save z-offset changed and saved (which you can confirm checking and comparing with files above) you can try printing any of the 1st layer test (best a small one so you don't waste material) and while it's printing live adjust (babystep) the offset with these buttons:

[image]

once you see a good layer you click save button in the same section that will become active when printing job is complete.

naveennisad, 2024-02-04:
1. rpi is running doesnt mean klipper is not restaring, any change that is not saved will be gone
2.  check if calculated zoffset is not overridden by some default value from config while printing, so everthing will go waste
3. I prefer putting caliberation in silcer start Gcode like below after setting bed and nozzle temp, and ensured nothing will override final calculation - 

M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X-2.0 Y20 Z0.28 F5000.0 ;Move to start position
M109 S[material_print_temperature_layer_0]

BED_MESH_CLEAR
PRTOUCH_PROBE_ZOFFSET
BED_MESH_CALIBRATE

pwlgrzs, 2024-02-04:
Putting z-offset calibration into start macro is VERY BAD advice, automated zoffset calibration must be supervised and confirmed visually, if something goes off and result is skewed for any reason you're risking nozzle nuking your bed.

VeGoVeVO, 2024-02-04:
> Putting z-offset calibration into start macro is VERY BAD advice, automated zoffset calibration must be supervised and confirmed visually, if something goes off and result is skewed for any reason you're risking nozzle nuking your bed.

thank you guys it works now i just saved the config file and flipped the bed now its printing perfectly i just want one more thing from anyone who knows '' when the printer starts a print first thing its doing is the nozzle cleaning process and from the looks of it you have a left side box that collects the debris right ? if so what if mine is to the right how can i change that because i have other things to the left. ''

naveennisad, 2024-02-04:
> Putting z-offset calibration into start macro is VERY BAD advice, automated zoffset calibration must be supervised and confirmed visually, if something goes off and result is skewed for any reason you're risking nozzle nuking your bed.

kind agree, that because I always monitor my printer fist layer start, with below checks -
is zoffset generated close to usual values?
how is generated mesh?
how is first layer?

if anything suspitious, ready to click emergency stop :)