0xD34D/klipper_ender3_v3_se #20: Bltouch is miles off

OldMateAidos, 2024-06-01:
Bltouch does not work properly constantly gives different offset  values Evan when done back to back , destroyed my print bed , ill stick to mainline klipper and use one firmware to proz z offset , thanks for ruing my pei bed

Atomique13, 2024-06-02:
That being your first issue on github says a lot.

OldMateAidos, 2024-06-03:
You can talk all the smack you want this is a horrible firmware , compiled mainline klipper and runs fines using stock fw your auto leveling the reading this firmware givea for zoffset are constanly off by 200 400

Atomique13, 2024-06-04:
> You can talk all the smack you want this is a horrible firmware , compiled mainline klipper and runs fines using stock fw your auto leveling the reading this firmware givea for zoffset are constanly off by 200 400

Lmao this being the first and only klipper firmware that has the working Auto Z Offset feature but okey. And as a matter of fact this firmware is being used by dozens of people without problems, but whatever rows your boat ig.

csutcliff, 2024-06-04:
PEBKAC

kultigspritzig, 2024-06-29:
I think I am having a similar issue: https://www.reddit.com/r/Ender3V3SE/comments/1drh1pk/is_my_probe_broken_or_what_prtouch_probe_zoffset/

Atomique13, 2024-06-29:
> I think I am having a similar issue: https://www.reddit.com/r/Ender3V3SE/comments/1drh1pk/is_my_probe_broken_or_what_prtouch_probe_zoffset/

Your sensor is most likely not working. You can try to install marlin and check with that too.

kultigspritzig, 2024-06-30:
> > I think I am having a similar issue: https://www.reddit.com/r/Ender3V3SE/comments/1drh1pk/is_my_probe_broken_or_what_prtouch_probe_zoffset/
> 
> Your sensor is most likely not working. You can try to install marlin and check with that too.

Yeah, that was it. Checked with stock firmware and it has same issue. The connectors are not loos, so it must have gone bust.