0xD34D/klipper_ender3_v3_se #25: MCU Protocol Error, mismatched klipper version when compiling firmware

Vib1240n, 2024-09-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Kiuah
2. Set custom repository to - https://github.com/0xD34D/klipper_ender3_v3_se
3. cd ~/klipper && make menuconfig
4. Set menuconfig options
5. Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32F103)  --->
    Bootloader offset (28KiB bootloader)  --->
    Communication interface (Serial (on USART1 PA10/PA9))  --->
6. Save and make
7. flash printer
**Expected behavior**
Flash the printer with updated firware

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser Zen browser, orcaSlicer
 - Version N/A

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
While compiling firmware I got this in the shell
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/dirzctl.o
  Compiling out/src/hx711s.o
src/hx711s.c: In function 'command_config_hx711s':
src/hx711s.c:55:5: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
   55 |     if (h->hx711_count > 4)
      |     ^~
src/hx711s.c:57:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
   57 |         h->flags = 0;
      |         ^
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_ldc1612.o
  Compiling out/src/sensor_hx71x.o
  Compiling out/src/sensor_ads1220.o
  Compiling out/src/sensor_bulk.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/stm32/clockline.o
  Compiling out/src/stm32/dfu_reboot.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/../lib/stm32f1/system_stm32f1xx.o
  Compiling out/src/stm32/adc.o
  Compiling out/src/stm32/stm32f1.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/stm32/i2c.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/stm32/serial.o
  Compiling out/src/generic/serial_irq.o
  Compiling out/src/stm32/hard_pwm.o
  Building out/compile_time_request.o
Version: 1a6c66ac
  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
  Creating hex file out/klipper.bin

whins, 2024-11-05:
I had same issue. I removed import prtouch.cfg and printer started working on this firmware

Grenfis, 2025-06-07:
> I was able to fix this issue by flashing stock firmware from creality, and then flashing klipper afterwards.

Thank you! I tried to install the firmware many times and every time failed, but your solution helped me!